zephyrproject-rtos/hal_espressif #34: ESP Log: Compilation error if CONFIG_LOG is not enabled.

b0661, 2021-07-05:
## Environment

- Development Kit:   Custom made
- Module or chip used:  ESP32-WROOM-32U
- hal-espressif: c21d252a2
- Zephyr: 91a6b286c9af2aad1ebbc6dbed72fcd149232b51 + PR#36378 + PR#36650 
- Build System:  Zephyr/CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:  Linux
- Using an IDE?: No
- Power Supply:  3.3V

## Problem Description

Compilation error if CONFIG_LOG is not enabled.

...components/log/include/esp_log.h:148:26: error: 'CONFIG_LOG_DEFAULT_LEVEL' undeclared (first use in this function); did you mean 'LOG_LOCAL_LEVEL'?

### Expected Behavior

Also work with CONFIG_LOG=n.

### Actual Behavior

See above.

### Steps to reproduce

Unset CONFIG_LOG or set CONFIG_LOG=n.

### Code to reproduce this issue

N/A

sylvioalves, 2021-07-06:
@b0661 I have tested same environment describe and it worked.  
I tested against `samples/boards/esp32/wifi_station`.  

Did you run the following commands?  
`west update` 
`west espressif update`

b0661, 2021-07-06:
Updated my project app code as a copy of  samples/boards/esp32/wifi_station (used to be hello app). 
Updated Zepyhr to: c3a067e84105bdd6bfcf0740bd5267e6120c98d5  + PR#36378 + PR#36650
Assured hal_espressif is uptodate (incl. submodules): c21d252a2188fcb60d52dceb6b018a9f4a371817  (hal: intc: add interrupt allocator hal support)

prj.conf: CONFIG_LOG=y
-> everything is fine

prj.conf: CONFIG_LOG=n

```
...
[68/322] Building C object zephyr/CMakeFiles/zephyr.dir/home/bobby/projects/comitberry/zephyr-hal-espressif/components/soc/src/esp32/rtc_clk.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/home/bobby/projects/comitberry/zephyr-hal-espressif/components/soc/src/esp32/rtc_clk.c.obj 
ccache /home/bobby/projects/comitberry/zephyr-sdk/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DBUILD_VERSION=zephyr-v2.6.0-751-g6867220e516a -DCONFIG_DPP -DCONFIG_ECC -DCONFIG_IEEE80211W -DCONFIG_SHA256 -DCONFIG_WPA3_SAE -DEAP_MSCHAPv2 -DEAP_PEAP -DEAP_PEER_METHOD -DESP32_WORKAROUND -DESPRESSIF_USE -DESP_PLATFORM -DESP_SUPPLICANT -DIEEE8021X_EAPOL -DKERNEL -DMBEDTLS_CONFIG_FILE=\"config-tls-generic.h\" -DUSE_WPA2_TASK -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -D__ets__ -I/home/bobby/projects/comitberry/comitberry-zephyr/kernel/include -I/home/bobby/projects/comitberry/comitberry-zephyr/arch/xtensa/include -I/home/bobby/projects/comitberry/comitberry-zephyr/include -Izephyr/include/generated -I/home/bobby/projects/comitberry/comitberry-zephyr/soc/xtensa/esp32 -I/home/bobby/projects/comitberry/comitberry-zephyr/soc/xtensa/esp32/include -I/home/bobby/projects/comitberry/comitberry-zephyr/subsys/net/l2 -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/include/bt -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/include/crypto -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/include/wifi -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/hal/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/hal/esp32/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_hw_support/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_hw_support/port/esp32 -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_hw_support/port/esp32/private_include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_common/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_rom/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_rom/esp32/ld -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp32/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/soc/esp32/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/xtensa/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/xtensa/esp32/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/soc/soc/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/soc/soc/esp32/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/soc/soc/esp32/private_include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/soc/include/soc -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/soc/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/soc/src/esp32/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/driver/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/soc/soc/esp32 -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_wifi/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/efuse/esp32/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/efuse/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_system/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_wifi/esp32/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_timer/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_timer/private_include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_netif/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/esp_event/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/log/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/wpa_supplicant/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/wpa_supplicant/port/include -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/wpa_supplicant/src -I/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/wpa_supplicant/include/esp_supplicant -I/home/bobby/projects [... cut ...]
/home/bobby/projects/comitberry/zephyr-hal-espressif/zephyr/esp32/../../components/log/include/esp_log.h:148:26: error: 'CONFIG_LOG_DEFAULT_LEVEL' undeclared (first use in this function); did you mean 'CONFIG_NET_IF_LOG_LEVEL'?
 #define LOG_LOCAL_LEVEL  CONFIG_LOG_DEFAULT_LEVEL
                          ^~~~~~~~~~~~~~~~~~~~~~~~
...
```

It is obvious that Zepyhr/Kconfig does not create CONFIG_LOG_DEFAULT_LEVEL in case CONFIG_LOG is not set.  esp_log.h needs CONFIG_LOG_DEFAULT_LEVEL in case LOG_LOCAL_LEVEL is not set. Do I have to set LOG_LOCAL_LEVEL somehow in my custom board environment (if so why)?

sylvioalves, 2021-07-07:
@b0661 ok, I was ble to reproduce now.  
I'll submit a PR soon. Find a fix below:

```c
diff --git a/components/log/include/esp_log.h b/components/log/include/esp_log.h
index 5a488936b0..88e336460a 100644
--- a/components/log/include/esp_log.h
+++ b/components/log/include/esp_log.h
@@ -144,12 +144,8 @@ void esp_log_writev(esp_log_level_t level, const char* tag, const char* format,
 #include "esp_log_internal.h"
 
 #ifndef LOG_LOCAL_LEVEL
-#ifndef BOOTLOADER_BUILD
-#define LOG_LOCAL_LEVEL  CONFIG_LOG_DEFAULT_LEVEL
-#else
 #define LOG_LOCAL_LEVEL  CONFIG_BOOTLOADER_LOG_LEVEL
 #endif
-#endif
 
 /** @endcond */
 
diff --git a/zephyr/esp32/include/sdkconfig.h b/zephyr/esp32/include/sdkconfig.h
index bf8e1825a4..df025bf352 100644
--- a/zephyr/esp32/include/sdkconfig.h
+++ b/zephyr/esp32/include/sdkconfig.h
@@ -1,5 +1,6 @@
 #define CONFIG_LOG_TIMESTAMP_SOURCE_RTOS 1
 #define CONFIG_ESP32_DPORT_DIS_INTERRUPT_LVL 5
+#define CONFIG_BOOTLOADER_LOG_LEVEL 3
 
 #define CONFIG_ESP32_PHY_MAX_TX_POWER 20
 #define CONFIG_ESP_MAC_ADDR_UNIVERSE_BT 1
 ```

b0661, 2021-07-07:
Thank you for the fast fix. It works.

closed 2021-07-07 by b0661