zephyrproject-rtos/hal_espressif #38: [ESP32] XTAL hardcoded to 40MHz.

cederom, 2021-07-09:
## Environment

- Development Kit:      [ESP32-SpartanEdgeAcceleratorBoard]
- Kit version: [v1]
- Module or chip used:  [ESP32-]
- IDF version (run ``git describe --tags`` to find it): `west espressif install` default proably `zephyr` head (`git describe --tags` does not work and there is no `west espressif info` to get any information`).
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp-2020r3) 8.4.0.
- Operating System:     [FreeBSD]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

XTAL frequency seems to be hardcoded to 40MHz instead AUTO. On this SEA board that I want to add to Zephyr XTAL of 26MHz needs to be used because Serial Console output is garbaged as described in [1]. How do I change this XTAL setting from Zephyr perspective? Can we set default to AUTO? Upstream `esp-idf` seems to have this problem solved already?

### Expected Behavior

* Serial Console output is garbaged.
* XTAL Configuration can be set to AUTO or changed from the Zephyr application standpoint.

### Actual Behavior

Only bootloader part serial console output is readable. Then output is garbaged.

### Steps to reproduce

1. Build Zephyr Hello World example.
2. See no expected output is produced but garbage.

### Code to reproduce this issue

Zephyr Hello World example.

## Debug Logs

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7752
load:0x40078000,len:13212
load:0x40080400,len:4568
entry 0x400806f4
W (65) rtc_clk_init: Possibly invalid CONFIG_ESP32_XTAL_FREQ setting (40MHz). Detected 26 MHz.
<EE><EE>^B<EC><EC><8C>ln<8C>b^\<E0>lb<8C><F2>nn<9E>^^l<90>l<80><92>b<8C>^P<E2>^B<8C><8C>l<9C><8E>
```

## Other items if possible

I have also opened a discussion for this issue at https://github.com/zephyrproject-rtos/zephyr/discussions/36827.

[1] https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-10-monitor

cederom, 2021-07-09:
Another description of this problem can be: `hal_espressif` does not respect `CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC` nor `clock-frequency` custom board definitions.

sylvioalves, 2021-07-09:
@cederom thanks for your comments and also regarding FreeBSD update. We'll submit a PR to add that.

1) Regarding changing XTAL value, this should be attached to your board design in .dtsi file:    
https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/xtensa/espressif/esp32.dtsi#L53  
Change it to `xtal-freq = <ESP32_CLK_XTAL_26M>;`  

2) We have not added yet the AUTO option. We will as well.  

3) Regarding CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC, it is a defconfig, currently set as 240MHz:  
https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/xtensa/esp32/esp32_defconfig#L10
If you want to change this value, you have to sync it with  
https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/xtensa/esp32/esp32.dts#L28  
otherwise it won't compile.  

Let me know if solves your issue.

cederom, 2021-07-09:
TANK U @sylvioalves YOU ROX! ITS ALIVE! =)

So it was `rtc` clock :-) All woks now as expected :-) `ESP32_CLK_XTAL_AUTO` would be safest and most universal choice :-)

sylvioalves, 2021-07-09:
Great! I have added this request into our feature development pipeline.

closed 2021-07-09 by sylvioalves

mbradley612, 2022-09-17:
I wonder if this bug has crept back in. 

Zephyr 3.2.0-rc2  with Zephyr SDK 0.15.0 on Ubuntu 22.04.1 LTS 

I am using a [Sparkfun ESP32 Thing](https://www.sparkfun.com/products/13907) development board which has a 26M crystal.

As described https://github.com/zephyrproject-rtos/zephyr/discussions/36827 I have set the following in the helloworld sample app 

zephyr/samples/hello_world/boards/esp32.overlay

```
&rtc {
    xtal-freq = <ESP32_CLK_XTAL_26M>;
};
```
When I run the monitor app, I get he following:
```
Detecting chip type... ESP32
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6596
load:0x40078000,len:15216
ho 0 tail 12 room 4
load:0x40080400,len:3992
entry 0x40080634
0x40080634: z_xtensa_backtrace_get_start at /home/matthew/zephyrproject/zephyr/arch/xtensa/core/debug_helpers_asm.S:29

W (61) rtc_clk_init: Possibly invalid CONFIG_ESP32_XTAL_FREQ setting (40MHz). Detected 26 MHz.
``` 
followed by garbage and then eventually:
```
*** Booting Zephyr OS build v3.2.0-rc2  ***n�#߇��<�b���
Hello World! esp32
```

Would it be possible to take another look at this please?

Many thanks.

sylvioalves, 2022-09-19:
@mbradley612, beside the garbage after Zephyr banner and the warning message, is it all working?