zephyrproject-rtos/hal_espressif #39: west flash --esp-device stored in configuration file

cederom, 2021-07-09:
Hello world :-)

Is it possible to store value of `west flash --esp-device` in some sort of configuration file so I don't have to type `west flash --esp-device=/dev/cuaU0` each time on my FreeBSD box? :-)

If this feature is not available can we please add this? :-)

Thank you :-)
Tomek

sylvioalves, 2021-08-02:
@cederom with this https://github.com/zephyrproject-rtos/zephyr/pull/37353 you will be able to export default port as in `export ESPTOOL_PORT=/dev/cuaU0` or even not defining the specific port using`--esp-device`. When not specifying the port, `esptool` will search for available serial ports and try to flash using it.

closed 2021-08-02 by sylvioalves

cederom, 2021-08-02:
Awsome! Thank you! :-)

ps/2: I have just ordered ESP32-C3 with RISC-V core - MINI and WROOM boards - will it work out of the box with Zpehyr? I am willing to help testing :-)