zephyrproject-rtos/hal_espressif #97: ESP32-S3 support

mbrunnen, 2022-03-09:
Hi, this a gently reminder that some of us are waiting desperately for S3 support :)

According to https://www.espressif.com/en/news/new_operating_systems_in_ESP32 it will be supported by the end of this month.

rftafas, 2022-03-09:
We are on it!

As stated on the link, we were going to **start** supporting it by Q1 and, actually, we are already doing so, but with some background tasks first (like AMP and MCUboot, both really important to make the most out of S3).

Going further on why AMP is important, ESP32S3 with a lone active core is 'cpu wise' pretty similar to ESP32S2. I also suggest reading Zephyr Bluetooth and SMP relation, to understand why AMP was our way to go.

GuptaAshi-Eaton, 2022-04-13:
@rftafas  do you have any timeline by when support for ESP32 S3 will be available in Zephyr as this will also need updates in ESP-IDF version that is currently supported in zephyr ?

rftafas, 2022-04-14:
Hi!

Status is the same as above. If you mean more direct action to support ESP32S3 (with CPU, basic set of peripherals and WiFi/BT), that is yet on the horizon. 

Yes, we will have to update hal_espressif to 4.4 (or later), based on ESP-IDF 4.4 (or later). Yep, this is a big hint we are approaching it.

---

In the meantime, we are suggesting developers to use ESP32-S2 (when USB is needed) or ESP32/ESP32-C3 (when BT is needed). Either on final solution or for the first prototyping round, depending on each particular reality.

Also thinking on prototype cycle... If it is mandatory to work with ESP32S3+Zephyr, I can suggest to give a try to the ESP-IDF/MCUboot that Amazon did and later migrate to Zephyr, as MCUboot currently supports ESP32S3 (quad flash only). It is undefined how much effort would be necessary. There could be dragons.

Reference: https://freertos.org/mcuboot/mcuboot-demo.html

GuptaAshi-Eaton, 2022-04-14:
> Hi!
>
> Status is the same as above. If you mean more direct action to support ESP32S3 (with CPU, basic set of peripherals and WiFi/BT), that is yet on the horizon.
>
> Yes, we will have to update [...]

Thanks for the response. Is there any timeline by when we can see ESP32-S3 supported on Zephyr ?

rftafas, 2022-04-14:
Unfortunately, not yet. There are too many steps ahead, a timeline would be wildly speculative.

CraigBurden, 2022-12-16:
Are there any updates on S3 support?

rftafas, 2022-12-16:
@CraigBurden It depends on point of view. We have been doing things that are actually part of our effort for S3 (check PRs on peripheral support, hal_espressif is 4.4 already). So, in a sense, lots of updates. As for more direct work on S3, meaning bring up, not yet. So, in other sense, not that much.

Same comment regarding ESP32S3 usage I made above is still valid. Depending on what you want to do, ESP32 is just as good.

CraigBurden, 2022-12-19:
@rftafas Thanks for the update, that is helpful. 

What I find confusing is that from my point of view the hold up is the implementation of AMP, being a hard requirement for multicore devices in Zephyr? But the ESP32 Vanilla is supported, which tells me the the IDF and the ESP32 can play together? And as you point out, the HAL is up to date? 

Honestly I'd be happy to use the S3 as a single core until this was all sorted. The S3 has many notable advantages over the S2, even when limping with only one core.

rftafas, 2022-12-23:
@CraigBurden 

> What I find confusing is that from my point of view the hold up is the implementation of AMP, being a hard requirement for multicore devices in Zephyr? But the ESP32 Vanilla is supported, which tells me the the IDF and the ESP32 can play together?

I assume it should be Zephyr+MP+Vanilla ESP32. It is confusing because that even on ESP32, you can't have SMP and Networking (check docs/do tests). That is the hard requirement. I suggest testing/verifying it yourself for the sake of exploration.

> And as you point out, the HAL is up to date?

Not the latest, but already contains S3 LL. No promises, though.

rftafas, 2023-01-02:
FWIW: https://github.com/zephyrproject-rtos/hal_espressif/pull/169