zephyrproject-rtos/hal_espressif #100: hal_espressif expects ZEPHYR_BASE environment variable to be set

stephanosio, 2022-03-21:
ESP32 targets do not build properly when the `ZEPHYR_BASE` environment variable is not set (i.e. when the Zephyr CMake package from the registry is used to find the Zephyr path).

```
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/devel/Dev/zephyrproject/zephyr/tests/kernel/threads/thread_apis/build/esp-idf/build/sdkconfig
Traceback (most recent call last):
  File "/home/devel/Dev/zephyrproject/modules/hal/espressif/tools/kconfig_new/confgen.py", line 42, in <module>
    sys.path.insert(0, os.path.join(os.environ['ZEPHYR_BASE'], "scripts", "kconfig"))
  File "/usr/lib/python3.8/os.py", line 675, in __getitem__
    raise KeyError(key) from None
KeyError: 'ZEPHYR_BASE'
CMake Error at /home/devel/Dev/zephyrproject/modules/hal/espressif/tools/cmake/kconfig.cmake:228 (message):
  Failed to run confgen.py
  (/usr/bin/python3.8;/home/devel/Dev/zephyrproject/modules/hal/espressif/tools/kconfig_new/confgen.py;--kconfig;/home/devel/Dev/zephyrproject/modules/hal/espressif/Kconfig;--sdkconfig-rename;/home/devel/Dev/zephyrproject/modules/hal/espressif/sdkconfig.rename;--config;/home/devel/Dev/zephyrproject/zephyr/tests/kernel/threads/thread_apis/build/esp-idf/build/sdkconfig;--env-file;/home/devel/Dev/zephyrproject/zephyr/tests/kernel/threads/thread_apis/build/esp-idf/build/bootloader/config.env).
  Error 1
Call Stack (most recent call first):
  /home/devel/Dev/zephyrproject/modules/hal/espressif/tools/cmake/build.cmake:451 (__kconfig_generate_config)
  /home/devel/Dev/zephyrproject/modules/hal/espressif/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:41 (project)
```

The `ZEPHYR_BASE` should be sourced from the `ZEPHYR_BASE` CMake variable as done here:
https://github.com/zephyrproject-rtos/zephyr/blob/70a1827fc1cbe65f5a7b8217f8d6259680d3974f/cmake/modules/kconfig.cmake#L156-L157

sylvioalves, 2022-04-29:
@stephanosio I could not reproduce this. Is it possible for you to re-check or share your env configurations regarding Zephyr and ESP?

henrikbrixandersen, 2022-05-18:
> @henrikbrixandersen So you just unset `ZEPHYR_BASE` from your env and build fails?

I normally never set `ZEPHYR_BASE` in the mirror environment; it should not be needed anymore.

henrikbrixandersen, 2022-05-18:
@sylvioalves Ah, it works out-of-the-box when using `west`, but not when using `cmake` + `make`.

I have submitted a fix here: https://github.com/zephyrproject-rtos/zephyr/pull/45773 (and here: https://github.com/zephyrproject-rtos/hal_espressif/pull/115).

sylvioalves, 2022-05-18:
Got it! Thanks.