zephyrproject-rtos/hal_espressif #203: pyserial dependency needs to be installed

beriberikix, 2023-03-25:
This is specific to the `west` integration and somewhat related to https://github.com/zephyrproject-rtos/zephyr/issues/56215.

The Espressif blob requires `pyserial` to be installed. However, it will only get installed if a user uses `requirements-run-test.txt` (not common) or gets ALL dependencies using `requirements.txt` per the Getting Started Guide. However, since pyserial isn't included in `requirements-base.txt`, anyone trying to maintain a lean build environment (like CI) will run into a dependency issue.

I'm not familiar with out `west blobs` work but is there a way to request additional dependencies?

sylvioalves, 2023-03-26:
I don't get why `west blobs` requires `pyserial`? Can you describe a little bit where it fails? Also, you can take a look at [west_cmds](https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/workflows/west_cmds.yml#L72), which is also used in CI and installs pyserial.

beriberikix, 2023-03-26:
@carlescufi asked to move to a new issue on the main repo, so I created #56224 and will close this one!

closed 2023-03-26 by beriberikix