zephyrproject-rtos/hal_espressif #272: zephyr: esptool too old?

jeru, 2024-02-12:
## Environment

- Development Kit:      ESP32-C3-SuperMini
- Kit version (for WroverKit/PicoKit/DevKitC): -
- Module or chip used:  ESP32-C3
- IDF version (run ``git describe --tags`` to find it):
    * IDF not involved (zephyr `west` based).
    * esptool version: v4.5
    * hal/espressif version after `west update`: (HEAD detached at [19d2fe44ba](https://github.com/zephyrproject-rtos/hal_espressif/commit/19d2fe44ba988781f83eb70645c3757f96fcdf1c))
- Build System:         zephyr `west`
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    riscv64-zephyr-elf-gcc.exe (Zephyr SDK 0.16.4) 12.2.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell.
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

Zephyr: `west build` + `west flash` failed.
BUT: took the failed command, replaces out the esptool path to the standard IDF/esptool, it works.
However, this replacement is too manual and hacky to do every time...

### Expected Behavior
`west build; west flash` should work.

### Actual Behavior
Failed to flash.

### Steps to reproduce

0. Update both clean `zephyr` and `modules/hal/espressif` git dirs with `west update`.
1. At zephyr git: `cd \samples\hello_world; west build -b esp32c3_devkitm .` Succeeded.
2. Then flashing: `cd build; west flash`  **Failed (and this is the problem)**. Full message:
```
-- west flash: rebuilding
[1/1] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Users\jerus...zephyr/zephyr/samples/hello_world/build/zephyr/zephyr.elf"
-- west flash: using runner esp32
-- runners.esp32: reset after flashing requested
-- runners.esp32: Flashing esp32 chip on None (921600bps)
esptool.py v4.5
Found 2 serial ports
Serial port COM11
Connecting...
COM11 failed to connect: Write timeout
Serial port COM1
Connecting......................................
COM1 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

A fatal error occurred: Could not connect to an Espressif device on any of the 2 available serial ports.
FATAL ERROR: command exited with status 2: 'C:\Users\jerus\zephyr\.venv\Scripts\python.exe' 'C:/Users/jerus/zephyr/modules/hal/espressif\components\esptool_py\esptool\esptool.py' --chip auto --baud 921600 --before default_reset --after hard_reset write_flash -u --flash_mode dio --flash_freq 40m --flash_size detect 0x0 C:/Users/jerus/zephyr/zephyr/samples/hello_world/build/esp-idf/build/bootloader/bootloader.bin 0x8000 C:/Users/jerus/zephyr/zephyr/samples/hello_world/build/esp-idf/build/partitions_singleapp.bin 0x10000 'C:\Users\jerus\zephyr\zephyr\samples\hello_world\build\zephyr\zephyr.bin'
```

3. HACK the failed command to:
```
C:/Embeddings/Espressif2/python_env/idf5.1_py3.11_env/Scripts/python.exe  C:/Embeddings/Espressif2/frameworks/esp-idf-v5.1.2/components/esptool_py/esptool/esptool.py --chip auto --baud 921600 --before default_reset --after hard_reset write_flash -u --flash_mode dio --flash_freq 40m --flash_size detect 0x0 C:/Users/jerus/zephyr/zephyr/samples/hello_world/build/esp-idf/build/bootloader/bootloader.bin 0x8000 C:/Users/jerus/zephyr/zephyr/samples/hello_world/build/esp-idf/build/partitions_singleapp.bin 0x10000 'C:\Users\jerus\zephyr\zephyr\samples\hello_world\build\zephyr\zephyr.bin'
```
(NOTE: I only changed the first two args, i.e., `python esptool` ).
It succeeds with message:
```
esptool.py v4.7.0
Found 2 serial ports
Serial port COM11
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: c0:4e:30:80:4d:ac
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00033fff...
Flash params set to 0x0220
Wrote 32768 bytes at 0x00000000 in 0.5 seconds (582.3 kbit/s)...
Hash of data verified.
Wrote 16384 bytes at 0x00008000 in 0.2 seconds (710.4 kbit/s)...
Hash of data verified.
Wrote 147456 bytes at 0x00010000 in 2.0 seconds (578.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

So it seems to me that esptool v4.5 carried with `zephyr` is tool old, while v4.7.0 is good enough.

EricNRS, 2024-02-12:
There is a big effort by the EspressIf team to upgrade the HAL to version 5.1 that is nearly complete which will resolve this issue.

jeru, 2024-02-12:
Any ETA known? And any suggestion to work around the issue temporarily? Thanks!

EricNRS, 2024-02-12:
> Any ETA known? And any suggestion to work around the issue temporarily? Thanks!

No ETA has been shared, yet, although the EspressIf team did share a link to the repos to start early testing.  Your temporary work-around of using esptool from IDFv5.1 is probably the best option short-term.

sylvioalves, 2024-05-02:
Already using latest version.

closed 2024-05-02 by sylvioalves